STMicroelectronics/STM32CubeU5 #3: Calling HAL_SPI_Transmit_DMA() generates crash with "half-duplex" SPI

SDS-BastienLS, 2022-01-11:
**Describe the set-up**

 * Nucleo-U575ZI-Q
 * STM32CubeIDE 1.7.0
 * Firmware Package version : V1.0.2
 * Toolchain (default STM32CubeIDE): GNU Tools for STM32 (9-2020-q2-update)

**Describe the bug (skip if none)**

HardFault on execution of this condition: 
Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_spi.c, line 2930

In function SPI2_IRQHandler()->HAL_SPI_IRQHandler()

After the call of HAL_SPI_Transmit_DMA().

**Additional context**

The code is generated with STM32CubeMX. 
The Cube project is configured with a SPI peripheral, and 1 DMA channel linked to the TX of the SPI.
Both IRQ Hanlder and HAL handler generation are ticked for both SPI and DMA channel interrupts.

The HAL_SPI_IRQ_Handler is trying to access the field "hspi->hdmarx->Mode" although hspi->hdmarx is NULL since no DMA channel is linked to SPI RX. This generates the HardFault.

ASELSTM, 2022-01-28:
ST Internal Reference: 121844

ASELSTM, 2022-03-11:
Hi @SDS-BastienLS,

This issue has been fixed in the frame of version v1.1.0 of the STM32CubeU5 published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM