STMicroelectronics/STM32CubeU5 #5: NUCLEO-U575ZI-Q / Ux_Device_CDC_ACM example:

djix123, 2022-02-07:
OS: OSX 12.1
IDE: STM32CubeIDE: 1.8.0 (fresh installation, all defaults)

Application builds with no errors, but when data transmission from USB->USART does not work correctly.
Only 1 character is transmitted, second and subsequent characters do not appear. By contrast, USART->USB works correctly - all characters are transmitted.

It appears that the call back ```HAL_UART_TxCpltCallback``` is not called on second and subsequent characters.

[Both the USB and USART via STLink VCP are connected to using ```screen```]

ALABSTM, 2022-05-09:
ST Internal Reference: 120699

ALABSTM, 2022-05-09:
Hi @djix123,

The issue you reported has been fixed in the frame of version **1.1.0** of this firmware, published some time ago. Please allow me to close this thread.

With regards,