STMicroelectronics/STM32CubeU5 #8: Typo in comment for DMA_CSR_SUSPF

SDS-BastienLS, 2022-03-28:
There is a typo in the comment, line 6621, file "Drivers\CMSIS\Device\ST\STM32U5xx\Include\stm32u575xx.h".

Original file:
```C
#define DMA_CSR_USEF_Msk                    (0x1UL << DMA_CSR_USEF_Pos)             /*!< 0x00001000 */
#define DMA_CSR_USEF                        DMA_CSR_USEF_Msk                        /*!< User setting error flag            */
#define DMA_CSR_SUSPF_Pos                   (13U)
#define DMA_CSR_SUSPF_Msk                   (0x1UL << DMA_CSR_SUSPF_Pos)            /*!< 0x00002000 */
#define DMA_CSR_SUSPF                       DMA_CSR_SUSPF_Msk                       /*!< User setting error flag            */
```

Should be:
```C
#define DMA_CSR_USEF_Msk                    (0x1UL << DMA_CSR_USEF_Pos)             /*!< 0x00001000 */
#define DMA_CSR_USEF                        DMA_CSR_USEF_Msk                        /*!< User setting error flag            */
#define DMA_CSR_SUSPF_Pos                   (13U)
#define DMA_CSR_SUSPF_Msk                   (0x1UL << DMA_CSR_SUSPF_Pos)            /*!< 0x00002000 */
#define DMA_CSR_SUSPF                       DMA_CSR_SUSPF_Msk                       /*!< Completed suspension flag          */
```

ASELSTM, 2022-03-29:
ST Internal Reference: 125355

ASELSTM, 2022-03-29:
Hi @SDS-BastienLS,

Thank you for this report. Indeed, you are right. The comment should be updated as you have mentioned. An internal bug tracker has been created and the fix will be made available in a future release.

With regards,