STMicroelectronics/STM32CubeU5 #11: Using FileX with STM32U5 internal nor flash

Nelthrion, 2022-06-20:
HI

I am using the STM32U575 for my project with the internal nor flash provided.
Noticing that the address is word(32bits) aligned while writing to the flash, but for STM32U5 internal flash the address needs to be quadword(128bits) aligned. Is there a configuration in levelx/filex (like a line size) with which levelx will provide me quadword aligned addresses, or I need to take care of this specifically in the flash write driver glue file(I think will affect the performance of write operation)?

Thanks

ASELSTM, 2022-06-24:
ST Internal Reference: 130915

ASELSTM, 2023-04-10:
Hi @Nelthrion,

Our development team has confirmed that it is possible to define the block size to 128bits. You might thus be able to develop such application.

With regards,

closed 2023-04-10 by ASELSTM