STMicroelectronics/STM32CubeU5 #17: stm32u5xx_hal_adc.h: `ADC4_RESOLUTION_XB` definitions

equationcrunchor, 2022-09-21:
In `stm32u5xx_hal_adc.h`, the `ADC4_RESOLUTION_XB` macros are under the `ADC_HAL_EC_RESOLUTION` group.  However, using these defines to configure ADC resolution with ADC4 does not lead to the correct resolution.  

Looking in `stm32u5xx_ll_adc.h`, we can see that the `LL_ADC_RESOLUTION_XB_ADC4` macros are labeled with a comment saying `/* Internal values only, please do not use */`.  But this same comment does not exist in `stm32u5xx_hal_adc.h`, which can lead to confusion.  

Comment should be present in both header files, and it should be clarified that the `ADC4_RESOLUTION_XB` macros cannot be used for ADC initialization.

PCHSTM, 2022-10-04:
Hello @equationcrunchor ,
Indeed this issue and confirmed and fix is under validation.
Fix details: merge all resolution settings in literals LL_ADC_RESOLUTION_xxB common to all ADC instances (remove specific literals LL_ADC_RESOLUTION_xxB_ADC4).
The fix will be available in STM32U5 FW package release v1.2.0 (in few weeks, beg. of November).

Best regards
Philippe

ALABSTM, 2022-10-04:
ST Internal Reference: 123906

ALABSTM, 2022-10-04:
Hi @equationcrunchor,

A few more details about the fix below, particularly how to update the `stm32u5xx_hal_adc_ex.c` file. Files `stm32u5xx_ll_adc.c` and `stm32u5xx_ll_adc.h` should also be updated (not shown here). As @PCHSTM wrote, the fix should be made available soon hopefully.

As for the [stm32u5xx_hal_adc_ex.c](https://github.com/STMicroelectronics/STM32CubeU5/blob/main/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_adc_ex.c) file, these lines should be updated as shown below:

https://github.com/STMicroelectronics/STM32CubeU5/blob/2f88245401997f2f4161724075721ae83980759b/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_adc_ex.c#L69-L76

```diff
-/* Values defined to be higher than worst cases: low clock frequency,         */
-/* maximum prescalers.                                                        */
-/* Ex of profile low frequency : f_ADC at 4,577 Khz (minimum value             */
-/* according to Data sheet), calibration_time MAX = 16384 / f_ADC             */
-/*           16384 / 4577.63671875 = 3.58s                                    */
-/* At maximum CPU speed (400 MHz), this means                                 */
-/*    3.58 * 400 MHz = 1432000000 CPU cycles                                  */
-#define ADC_CALIBRATION_TIMEOUT         (1432000000U)   /*!< ADC calibration time-out value */
+/* Values defined to be higher than worst cases: maximum ratio between ADC    */
+/* and CPU clock frequencies.                                                 */
+/* Example of profile low frequency : ADC frequency minimum 140kHz (cf        */
+/* datasheet for ADC4), CPU frequency 160MHz.                                 */
+/* Calibration time max = 25502 / fADC (refer to datasheet)                   */
+/*                      = 29M CPU cycles                                      */
+#define ADC_CALIBRATION_TIMEOUT         (29000000U)   /*!< ADC calibration time-out value */
```

With regards,

ASELSTM, 2023-04-06:
Hi @equationcrunchor,

Thank you for your contribution. This issue has been fixed in the frame of version v1.2.0 of the STM32CubeU5. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM