STMicroelectronics/STM32CubeU5 #20: Blocking issue with power consumption demo PWR_ModesSelection

bigbearishappy, 2022-11-08:
**Describe the set-up**

 * The board
NUCLEO-U575ZI-Q
 * IDE or at least the compiler and its version
STM32CubeIDE
Version: 1.10.1
Build: 12716_20220707_0928 (UTC)
compiler:
gcc 10.3-2021.10

**Describe the bug (skip if none)**
The board can not enter the relative low power mode.

**How to reproduce the bug**
1 remove the JP5 and connect an amperemeter to it
1 compile and download the demo code under Projects\NUCLEO-U575ZI-Q\Examples\PWR\PWR_ModesSelection
2 config the low power via the usart
3 finish the config and enter the low power mode
4 in STOP0 mode the current consumption should be about 39uA，but it's about 7mA

**Additional context**
I did some work on tracing the issue. And I find the program blocked after run into the function [system_config](https://github.com/STMicroelectronics/STM32CubeU5/blob/2e2b3e4733ddcde7af1e91f1c27ef7bea84389ce/Projects/NUCLEO-U575ZI-Q/Examples/PWR/PWR_ModesSelection/Src/lowpower_scenarios.c#L182)

I hope someone can help me out~

bridadan, 2022-12-13:
I can confirm I'm having the same issue. I'm seeing a ~7.5mA current draw no matter the PWR mode I select.

HBOSTM, 2022-12-21:
Hello @bigbearishappy
We confirm the abnormal behavior you reported. It has been reported to our development teams for deeper analysis. We will get back to you as soon we have updates.

With regards,

HBOSTM, 2022-12-21:
ST Internal Reference: 141762

LaVoute974, 2023-03-20:
Hello, I'm having the same issue (consumption arround 5mA in every mode), do you have any updates on this ?

HBOSTM, 2023-04-04:
Hi @LaVoute974 ,

Thank you for your contribution. This issue has been fixed. Could you please try with the last version **v1.2.0**.

Best Regards,

closed 2023-04-14 by RJMSTM