STMicroelectronics/STM32CubeU5 #23: stm32u5a9j-dk - MX_USB_OTG_HS_PCD_Init() timeout

AndrewCapon, 2023-03-14:
Hi Guys,

To replicate:

Use STM32CubeIde (1.12.0) to generate a new project for this board and leave everything as default, generate the code and run it.
We have host usb and the call to MX_USB_OTG_HS_HCD_Init() works correctly.

Now edit the IOC and change USB_OTG_HS Internal HS Phy from host_only to device_only and regenerate the code and run it.
Now we have device usb and the call to  MX_USB_OTG_HS_PCD_Init() times out in USB_CoreReset().

I notice there are no device based examples either, is the device side of things not working yet?

AndrewCapon, 2023-03-15:
Hi Guys,

This all seems to be to do with the clocks maybe, I could not get it going with the default clock settings.

I used the clock generator to set to 160mhz, set SDMMC mux to clk48 and set OTG HS mux to HSE.

I then added `__HAL_RCC_SYSCFG_CLK_ENABLE();` to `HAL_MspInit()` :

`void HAL_MspInit(void)
{
  /* USER CODE BEGIN MspInit 0 */
  __HAL_RCC_SYSCFG_CLK_ENABLE();
  /* USER CODE END MspInit 0 */

  __HAL_RCC_PWR_CLK_ENABLE();

  /* System interrupt init*/

  /* USER CODE BEGIN MspInit 1 */

  /* USER CODE END MspInit 1 */
}`

Now it isn't timing out.

fyyxxm, 2023-10-26:
I have also encountered this issue, as both running official routine code and code generated using CUBEMX will result in a USB_CoreReset timeout.

ALABSTM, 2024-02-06:
Hi @AndrewCapon,

Thank you for this report and analysis. When you speak about "editing the IOC", is it manually or via the `STM32CubeMX` tool? Thank you in advance for your answer.

With regards,

ALABSTM, 2024-02-06:
Hi @AndrewCapon,

After browsing into the different applications proposed within the firmware, I noticed the call to `__HAL_RCC_SYSCFG_CLK_ENABLE()` is systematic whether the USB is in host mode or in device more. Below some examples.

This looks an issue related to the `STM32CubeMX` tool that does not generate the required macro invocation. It has been forwarded to our development teams. A fix should be available in a future release.

As the issue is not related to the software provided within this repository, please allow me to close it. Thank you for your comprehension and thank you again for having reported.

With regards,

https://github.com/STMicroelectronics/STM32CubeU5/blob/c8fcb26ff629cf7c2a3b2c60e6121625eaa5ca2d/Projects/STM32U5x9J-DK/Applications/USBX/Ux_Host_HUB_HID_MSC/Core/Src/stm32u5xx_hal_msp.c#L161-L170

https://github.com/STMicroelectronics/STM32CubeU5/blob/c8fcb26ff629cf7c2a3b2c60e6121625eaa5ca2d/Projects/STM32U5G9J-DK2/Applications/USBX/Ux_Device_CDC_ACM/Core/Src/stm32u5xx_hal_msp.c#L198-L207

closed 2024-02-06 by ALABSTM

ALABSTM, 2024-02-06:
ST Internal Reference: 172800