STMicroelectronics/STM32CubeU5 #28: TFM_Appli directory missing in B-U585I-IOT02A TFM Application 1.2.0

rwsanders, 2023-06-30:
The "Directory Contents"section of the readme states that the directory `TFM_Appli` should exist. When I pull and build this project it fails because this directory does not exist. 

[STM32CubeU5-TFM](https://github.com/STMicroelectronics/STM32CubeU5/tree/v1.2.0/Projects/B-U585I-IOT02A/Applications/TFM)

RJMSTM, 2023-07-04:
Hello @rwsanders,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

TOUNSTM, 2023-07-11:
Hello @rwsanders,

Some projects have been made unavailable in our repositories on GitHub as they are subject to "click-thru" licenses.
If needed, the removed items can be found in the full firmware package on st.com. However, to avoid any misunderstood, we will update our documentation to highlight this point.

Please allow me thus to close this thread. Thank you for you comprehension.

With Regards,

closed 2023-07-11 by TOUNSTM