STMicroelectronics/STM32CubeU5 #31: stm32u5a9xx.h USB_OTG mess

tdjastrzebski, 2023-09-19:
1. File [stm32u5a9xx.h](https://github.com/STMicroelectronics/STM32CubeU5/blob/main/Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5a9xx.h), line 27339:
`#define USB_OTG_GCCFG_FSVMINUS_Msk 0x1U << USB_OTG_GCCFG_FSVMINUS_Pos) /*!< 0x00000004 */`
has missing opening parenthesis
2. GINTSTS.RSTDET registry field definition (`USB_OTG_GINTSTS_RSTDET`) is missing.
See: RM0456, 73.14.6 OTG core interrupt register [alternate] (OTG_GINTSTS)
3. Many USB_OTG registry field names differ from those described in RM0456. Some of them are completely different. Examples:
GINTSTS.WKUPINT -> USB_OTG_GINTSTS_WKUINT
GINTSTS.IPXFR -> USB_OTG_GINTSTS_PXFR_INCOMPISOOUT
GINTSTS.GONAKEFF -> USB_OTG_GINTSTS_BOUTNAKEFF
GCCFG.FORCEHOSTPD -> USB_OTG_GCCFG_PULLDOWNEN
GCCFG.VBVALOVEN -> USB_OTG_GCCFG_VBVALEXTOEN
GCCFG.HVDMSRCEN -> USB_OTG_GCCFG_H_VDMSRCEN
GCCFG.HCDPDETEN -> USB_OTG_GCCFG_H_CDPDETEN
GCCFG.HCDPEN -> USB_OTG_GCCFG_H_CDPEN
4. OTG_HCFG register (RM0456, 73.14.24 OTG host configuration register)  at offset 0x400 is not defined.
5. OTG_HPRT register (RM0456, 73.14.30 OTG host port control and status register) at offset  0x440 is not defined

It is hard to avoid impression that this STM32U5 HAL release was prepared in rush and corners were cut.

RJMSTM, 2023-09-20:
Hello @tdjastrzebski,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

tdjastrzebski, 2023-09-21:
Hello @RJMSTM. I just discovered that more USB_OTG register definitions are missing.

- GOTGCTL.CURMOD
- GOTGCTL.OTGVER
- GOTGCTL.ASVLD
- GOTGCTL.DBCT
- GOTGCTL.CIDSTS
- GOTGCTL.EHEN
- and  GOTGCTL.BSVLD is incorrectly named USB_OTG_GOTGCTL_BSESVLD

See [RM0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf),  73.14.1 OTG control and status register (OTG_GOTGCTL)

ALABSTM, 2024-01-12:
ST Internal Reference: 170735

ALABSTM, 2024-01-12:
Hi @tdjastrzebski,

Please excuse this delayed reply. Thank you very much for this detailed report. It has been forwarded to our development teams. I will keep you informed.

With regards,

ALABSTM, 2024-01-12:
ST Internal Reference: 170738

ALABSTM, 2024-01-12:
ST Internal Reference: 170739

ALABSTM, 2024-01-25:
ST Internal Reference: 171759

TOUNSTM, 2024-04-05:
Hello @tdjastrzebski,

> 1. File [stm32u5a9xx.h](https://github.com/STMicroelectronics/STM32CubeU5/blob/main/Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5a9xx.h), line 27339:
>    `#define USB_OTG_GCCFG_FSVMINUS_Msk 0x1U << USB_OTG_GCCFG_FSVMINUS_Pos) /*!< 0x00000004 */`
>    has missing opening parenthesis!

The issue you have raised has been fixed in commit 332197d7ed2d96a9ee3aaf824181c5a3bb3541cb.

With Regards,

ALABSTM, 2024-10-04:
Hi @tdjastrzebski,

We're almost there to close this issue:
* Point 1 has already been fixed in commit 2e053bcab594083bc519a69fbc72bb95d57881db.
* Points 2 and 3 will be fixed in the frame of release **1.7.0**, coming in a couple of weeks hopefully.
* Point 4, however, has been declined by our development teams.

I will keep you informed once release **1.7.0** becomes available on GitHub.

With regards,

tdjastrzebski, 2024-10-05:
Hi @ALABSTM, thank you for the update. nice It is good to see that ST software catches up with hardware. Batter late than never. Seriously.

TOUNSTM, 2024-11-28:
Hello @tdjastrzebski,

The points 2 and 3 have already been fixed in the frame of [release 1.7.0 ](https://github.com/STMicroelectronics/STM32CubeU5/commit/7392378474639cdcd87a71918f5272f847454055).

Thank you for your valuable contribution. We are looking forward to reading from you again.
With Regards,

closed 2024-11-28 by TOUNSTM