STMicroelectronics/STM32CubeU5 #34: Nucleo U5A5ZJ-Q USBX - Ux_Device_CDC_ACM MX_USB_OTG_HS_PCD_Init() timeout with MSIS

darshanmirajkar-eaton, 2023-12-15:
Trying to run STM32CubeU5\Projects\NUCLEO-U5A5ZJ-Q\Applications\USBX\Ux_Device_CDC_ACM of **main** branch with HSI facing the issue of MX_USB_OTG_HS_PCD_Init() timeout with MSIS

 * Nucleo U5A5ZJ-Q
 * IDE version 1.13.2

MX_USB_OTG_HS_PCD_Init() timeout in USB_CoreReset.

**Steps to reproduce**

1. Open IOC. Disable HSE
[image]

2. Set the clock as shown in image
[image]
[image]

3. While initializing MX_USB_OTG_HS_PCD_Init() timeout can be observed while debugging
[image]

Can someone help to identify issue and resolve this.

ALABSTM, 2023-12-19:
Hi @darshanmirajkar-eaton,

From what I understood of the modification you brought to the project is that your system will start being clocked by the `MSI` clock **just after the reset**.

In the screenshots you shared, the `PLL1P` clock, derived from the `MSI` clock, is indicated to be oscillating at 20MHz. However, according to page **486** of the [RM0456 Rev.4](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf) "After restart from reset or when exiting Shutdown mode, MSIS 
and MSIK frequencies are set to their default value **4 MHz**".

Could this explain the timeout you are experiencing? Could it be that you should leave the `HSE` clock enabled after a reset and switch to the `MSI` clock **only after** ensuring this latter is oscillating at the desired frequency, **i.e.**, 20MHz after it has been configured to? Please keep me informed.

With regards,

darshanmirajkar-eaton, 2023-12-19:
@ALABSTM I tried HSE enabled after reset and switch to MSI. This case I am not experiencing time out. 
But in our case, we are not using HSE. USB should work with MSI also, right??

darshanmirajkar-eaton, 2023-12-20:
@ALABSTM  I further investigated.
I Tried this OTG HS Clock Mux with HSE is working. OTG HS Clock Mux with PLL1P leads to MX_USB_OTG_HS_PCD_Init() timeout in USB_CoreReset.
[image]

[image]

ALABSTM, 2023-12-21:
Hi @darshanmirajkar-eaton,

From what I see from your last test, although your USB OTG HS clock is fed by the HSE clock, your PLL1P is still fed by the MSIS. Hence, the same issue and the timeout.

According to our development teams, the recommendation is to use the HSE clock for the USB OTG HS. An alternative is the HSE clock bypass mode using an external clock source. This would allow you to use only the `OSC_IN` pin and spare the `OSC_OUT`. You can refer to the reference manual for more details.

I am not sure whether the HSI clock would be any of a solution. However, our development teams strongly discourage such option. I hope this helps.

With regards,

darshanmirajkar-eaton, 2023-12-22:
@ALABSTM USB OTG is working with HSE. Is there any specific reason not to use PLL1P? 

And another observation is with PLL1P/2 at least I am not observing timeout, but USB is not working.

tdjastrzebski, 2024-01-06:
@darshanmirajkar-eaton Check out this post [Why OTG_HS USB on STM32U5A9x just does not work?](https://community.st.com/t5/stm32-mcus-products/why-otg-hs-usb-on-stm32u5a9x-just-does-not-work/td-p/592702)
It seems that indeed there is some hardware bug that ST staff is not allowed to admit (no offence, I do understand).

tdjastrzebski, 2024-01-08:
@darshanmirajkar-eaton Note that MSIS clock may not be correctly synced with LSE unless RTC is enabled as well. There was a bug in CubeMX I reported [here](https://community.st.com/t5/stm32-mcus-products/crs-sync-setup-with-lse-and-cubemx-puzzle/td-p/604835), it seems it was corrected in the latest release 6.10.

darshanmirajkar-eaton, 2024-01-10:
@tdjastrzebski we able to initialize using MSIS with below clock settings  
[image]
[image]
Working only with PLL1P/2. With PLL1P still facing timeout issue

tdjastrzebski, 2024-01-10:
@darshanmirajkar-eaton Thank you for the confirmation, it is consistent with my observations: HSE or "/2" must be used.

riscy00, 2024-01-21:
I have the same USB core reset issue with the STM325UA5RJ nucleo board with V140 CubeMX (CubeIDE is V141, just updated). I generated the code from the start via CubeMX on CDC_ACM (Device). I use HSE of 16MHz that feeds to the USB_PHY device. I get CSRST not clearing itself. After reading this comment here, I'm unsure what needs to be done to fix this; thanks. The older demo code (dated 2021) CDC_ACM is working fine but I cannot use this version.

riscy00, 2024-01-21:
I have tried MSIS, HSE and RTC (LSE) setting variants as described above and found not working with CSRTS self clearing and time out.

tdjastrzebski, 2024-01-22:
There are two problems:
1. For the IP clock HSE or PLL1P **with /2 divider** and 2x higher frequency has to be used. PLL1P without divider does not work. Why? I have not clue, but it seems like it is a hardware bug ST is not ready to confirm yet.
2.  CRS SYNC did not work (was not really enabled), unless RTC was enabled as well. I have reported this CubeMX bug to ST, it has been fixed in the latest release 6.10.

riscy00, 2024-01-22:
I have not tried working the solution in linked https://github.com/tdjastrzebski/Nucleo-U5A5ZJ-USBX since it is set up as Host. I am currently on Device under CDC_ACM under 1.4.0/1.4.1 CubeMX/IDE install. Does anyone have a working USB / USBX device (regardless of whether this is CDC_ACM or variants)? I do not wish to use the demo version supplied by STM since it was written in 2021, and I do not want to downgrade to an older version of Cube IDE/MX in case of other bugs in Azure stuff and library.

riscy00, 2024-01-22:
I have placed the code __HAL_RCC_SYSCFG_CLK_ENABLE(), as shown below, and the CSRST issue went away, but I do not see the COM port on my laptop. It seems the config clock was left disabled before configuring. I review it later in the week and maybe make a fresh Device CDC_ACM project from Cube IDE/MX V140/V141. I have applied PLL1P/2 and RTC enabled.

void HAL_HCD_MspInit(HCD_HandleTypeDef* hhcd)

{

  RCC_PeriphCLKInitTypeDef PeriphClkInit = {0};

  if(hhcd->Instance==USB_OTG_HS)

  {

  /* USER CODE BEGIN USB_OTG_HS_MspInit 0 */

  __HAL_RCC_SYSCFG_CLK_ENABLE();

  /* USER CODE END USB_OTG_HS_MspInit 0 */

From Palm LLM: The HAL_RCC_SYSCFG_CLK_ENABLE() function enables the clock for the System Configuration Controller (SYSCFG) in STM32U5 microcontrollers. The SYSCFG peripheral is responsible for configuring and managing various system-level features such as power management, clock control, and interrupt handling. By enabling the SYSCFG clock, you can access and control these features through the HAL library functions.

darshanmirajkar-eaton, 2024-01-24:
Can you share the clock settings @riscy00

riscy00, 2024-01-30:
[046_STMU5A5_AzureUSBX.txt](https://github.com/STMicroelectronics/STM32CubeU5/files/14102301/046_STMU5A5_AzureUSBX.txt)
Rename the postfix to ioc. This contains the clock setting. The USPD is not generating interrupts to start the USB so it remains disconnected from USB bus. I think there bug in USPD or incomplete code on V140/V141 library. The missing source code that process CAD which make it hard to figure out just cause. Can you supply USBPD_CAD_Process() source code?

ALABSTM, 2024-02-06:
Hi everyone,

Thank you for this instructive exchange. Glad to read you found a solution.

As this is not related to the software published within this repository, but rather to hardware aspects, please allow me to close this thread. Nevertheless, you can continue using it to exchange further about this topic. Thank you for your comprehension.

With regards,

tdjastrzebski, 2024-02-06:
@ALABSTM, @riscy00 Indeed, so far everything suggest it is a hardware issue. I believe ST is in the process of analyzing it.

ALABSTM, 2024-02-06:
Hi everyone,

Further details from the reference manual [RM0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf) Rev.5, page `487`, paragraph `11.4.3 MSI (MSIS and MSIK) clocks`:

> The MSI advantage is to provide a low-cost (no external components) low-power clock source. In addition, when used in PLL-mode with the LSE, the MSI provides a very accurate clock source that can be used by the OTG_FS, or the USB, and feeds the PLL.

Thank you all again.

tdjastrzebski, 2024-02-06:
@ALABSTM One comment: RM0456 p. 487 mentions OTG_FS, but not OTG_HS. Maybe it applies to OTG_HS as well, but maybe not.

ALABSTM, 2024-02-06:
@tdjastrzebski,

Thank you for your comment. I updated mine above. To conclude:
* There is **no** hardware bug.
* The OTG HS, to be correctly used, needs the `HSE` clock as clock source or the `PLL1 P` **clocked by** the `HSE` clock.

Other sources, like the `MSI` clock shall **not** be used with the OTG HS.

With regards,

tdjastrzebski, 2024-02-06:
@ALABSTM I think it may be a good idea to update RM0456. The only OTG_HS clock requirement mention is 400 ppm accuracy, which, according to the actual measurements I did, can be fulfilled using LSE sync-ed MSIS clock.
However, I understand that it is more than that and even LSE sync-ed MSIS clock is not good enough to be used for OTG_HS.
Needless to say, It is not a good news because that means that MCUs equipped with OTG_HS require LSE. no matter if you use high speed or not. It is a disadvantage.