STMicroelectronics/STM32CubeU5 #37: TFM Application does not build

tdjastrzebski, 2024-01-09:
According to the `readme.html` found in TFM application root folder:

_In order to build the TFM_Appli Secure and Non Secure projects, you must do the following:
• Open your preferred toolchain
• Rebuild the two projects_

I have tried, but:
1. Application does not build. All build did was it created lots of **yaml** files.
2. Solution structure nowhere reassembles what is documented in [UM2851 - Getting started with STM32CubeU5 TFM application](https://www.st.com/resource/en/user_manual/dm00778528-.pdf), page 33 - screenshot below

CubeMX and CubeIDE are the latest version

Please advise.

[image]

[image]

[image]

RJMSTM, 2024-01-11:
Hello,

Actually this application has been removed as indicated in the [README.md](https://github.com/STMicroelectronics/STM32CubeU5?tab=readme-ov-file#some-middleware-libraries-and-projects-are-unavailable-in-this-repository) file. This is due to licensing aspects preventing from exposing it on GitHub and  now is out of the scope of our GitHub' activity. They are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. Please allow me to redirect you to post your request in the adequate topic from [this](https://community.st.com/s/topiccatalog) list.

Now, as this issue is not directly related to some software published in GitHub, please allow me to close it.

More details can be found in https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17 post.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

RJMSTM, 2024-01-11:
Hello @tdjastrzebski ,

Actually this application has been removed as indicated in the [README.md](https://github.com/STMicroelectronics/STM32CubeU5?tab=readme-ov-file#some-middleware-libraries-and-projects-are-unavailable-in-this-repository) file. This is due to licensing aspects preventing from exposing it on GitHub and now is out of the scope of our GitHub' activity. They are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. Please allow me to redirect you to post your request in the adequate topic from [this](https://community.st.com/s/topiccatalog) list.

Now, as this issue is not directly related to some software published in GitHub, please allow me to close it.

More details can be found in https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17 post.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

closed 2024-02-05 by ALABSTM