STMicroelectronics/STM32CubeU5 #38: SBSFU application does not build

tdjastrzebski, 2024-01-09:
SBSFU Application does not build. According to the readme.html found in SBSFU application root folder:

_In order to build the SBSFU_Appli Secure and Non Secure projects, you must do the following:
• Open your preferred toolchain
• Rebuild the two projects_
Binary builds but then `postbuild.sh` scripts looks for some nonexistent paths and `C:\scripts\\imgtool\main.py`.
Was it tested by anyone before publishing?
CubeMX and CubeIDE are the latest version

Please advise.

[image]
[image]

RJMSTM, 2024-01-11:
Hello @tdjastrzebski,

Actually, this application has been removed as indicated in the [README.md](https://github.com/STMicroelectronics/STM32CubeU5?tab=readme-ov-file#some-middleware-libraries-and-projects-are-unavailable-in-this-repository) file. This is due to licensing aspects preventing from exposing it on GitHub and now is out of the scope of our GitHub' activity. They are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. Please allow me to redirect you to post your request in the adequate topic from [this](https://community.st.com/s/topiccatalog) list.

Now, as this issue is not directly related to some software published in GitHub, please allow me to close it.

More details can be found in https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17 post.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

RJMSTM, 2024-01-11:
Hello @tdjastrzebski ,

Actually this application has been removed as indicated in the [README.md](https://github.com/STMicroelectronics/STM32CubeU5?tab=readme-ov-file#some-middleware-libraries-and-projects-are-unavailable-in-this-repository) file. This is due to licensing aspects preventing from exposing it on GitHub and now is out of the scope of our GitHub' activity. They are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. Please allow me to redirect you to post your request in the adequate topic from [this](https://community.st.com/s/topiccatalog) list.

Now, as this issue is not directly related to some software published in GitHub, please allow me to close it.

More details can be found in https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/17 post.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

tdjastrzebski, 2024-01-12:
@RJMSTM Not true, SBSFU application has NOT been removed. It can be found here:
https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/SBSFU
Moreover, SBSFU is not listed here:
https://github.com/STMicroelectronics/STM32CubeU5/blob/main/Projects/README.md#list-of-unavailable-projects
What was removed was TFM application. FYI: TFM and SBSFU are two different applications.
The first one uses ARM TF-M while the other, well, officially is not.