STMicroelectronics/STM32CubeU5 #39: FileX boot record, ensuring data persistence

Dadigno, 2024-01-15:
I am currently developing on a custom board based on the STM32U5 microcontroller with the MX25UM512 NOR flash memory. 

I am following the [example](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/STM32U5x9J-DK/Applications/FileX/Fx_NoR_Write_Read_File) provided by the STM32CubeU5 SDK. The example works correctly when I first run it (so data is written correctly). However, upon subsequent runs following a reset, the previously written data seems to vanish, as a new "STM32.TXT" file is created without FX_ALREADY_CREATED

After conducting a thorough investigation, I discovered that the fx_media_format overwrites the boot record during each invocation. To avoid this I modified the example code as follow.

```
void fx_app_thread_entry(ULONG thread_input)
 {

  UINT nor_ospi_status = FX_SUCCESS;

/* USER CODE BEGIN fx_app_thread_entry 0*/
	  ULONG bytes_read;
	  ULONG available_space_pre;
	  ULONG available_space_post;
	  CHAR read_buffer[32];
	  CHAR data[] = "This is FileX working on STM32";

	  //printf("FileX/LevelX NOR OCTO-SPI Application Start.\n");

	  /* Print the absolute size of the NOR chip*/
	  //printf("Total NOR Flash Chip size is: %lu bytes.\n", (unsigned long)LX_STM32_OSPI_FLASH_SIZE);

/* USER CODE END fx_app_thread_entry 0*/

  /* Open the OCTO-SPI NOR driver */

 nor_ospi_status =  fx_media_open(&nor_ospi_flash_disk, FX_NOR_OSPI_VOLUME_NAME, fx_stm32_levelx_nor_driver, (VOID *)LX_NOR_OSPI_DRIVER_ID, (VOID *) fx_nor_ospi_media_memory, sizeof(fx_nor_ospi_media_memory));

/* Check the media open nor_ospi_status */
  if (nor_ospi_status != FX_SUCCESS)
  {
    /* USER CODE BEGIN OCTO-SPI NOR open error */

	  
	    nor_ospi_status =  fx_media_format(&nor_ospi_flash_disk,                                                           // nor_ospi_flash_disk pointer
	                                       fx_stm32_levelx_nor_driver,                                                     // Driver entry
	                                       (VOID *)LX_NOR_OSPI_DRIVER_ID,                                                  // Device info pointer
	                                       (UCHAR *) fx_nor_ospi_media_memory,                                             // Media buffer pointer
	                                       sizeof(fx_nor_ospi_media_memory),                                               // Media buffer size
	                                       FX_NOR_OSPI_VOLUME_NAME,                                                        // Volume Name
	                                       1,                                                     // Number of FATs
	                                       32,                                                                             // Directory Entries
	                                       0,                                                     // Hidden sectors
	                                       (LX_STM32_OSPI_FLASH_SIZE - LX_STM32_OSPI_SECTOR_SIZE)/ FX_NOR_OSPI_SECTOR_SIZE,// Total sectors
	                                       FX_NOR_OSPI_SECTOR_SIZE,                                                        // Sector size
	                                       8,                                                                              // Sectors per cluster
	                                       1,                                                                              // Heads
	                                       1);                                                                             // Sectors per track

	    if (nor_ospi_status != FX_SUCCESS)
	    {
	      while(1);
	    }

	    
	    nor_ospi_status =  fx_media_open(&nor_ospi_flash_disk, FX_NOR_OSPI_VOLUME_NAME, fx_stm32_levelx_nor_driver, (VOID *)LX_NOR_OSPI_DRIVER_ID, (VOID *) fx_nor_ospi_media_memory, sizeof(fx_nor_ospi_media_memory));
	    if (nor_ospi_status != FX_SUCCESS)
	    {
	    	while(1);
	    }
    /* USER CODE END OCTO-SPI NOR open error */
  }

/* USER CODE BEGIN fx_app_thread_entry 1*/

	/* Get the available usable space */
	nor_ospi_status =  fx_media_space_available(&nor_ospi_flash_disk, &available_space_pre);
......
...
.
```
This code, previously checks if media can be opened without problems, if not formatting is executed than re-opened.

However, data is not maintained, which I think is because the boot record isn't updated after file creation. So, after a reset, when sector 0 is read, the STM32.TXT file is gone.

I am not sure why this is happening. I would appreciate any help you can provide.

TOUNSTM, 2024-09-04:
Hello @Dadigno,

Thank you for your contribution. I think your issue has been resolved in the frame of the  [Release v1.5.0](https://github.com/STMicroelectronics/STM32CubeU5/commit/5ad32958b27c2d447a9438503efa6c2d0ccd8030). Could you please try the latest version and confirm if the problem is resolved?

Thanks again for your report.

TOUNSTM, 2024-10-14:
Hello @Dadigno,

Thank you once again for your contribution. Since we have not received any further response, we will close this thread for now. If you encounter any more issues or have additional details to provide, please feel free to reopen the thread at any time.

Thank you for your understanding.
Best Regards,

closed 2024-10-14 by TOUNSTM