STMicroelectronics/STM32CubeU5 #42: CRYP Key & IV configuration does not respect configured DataType

fermentedfly, 2024-02-29:
stm32u5xx_hal_cryp

We use AES in byte mode (`CRYP_DATATYPE_8B`).

CRYP_ConfigTypeDef provides the option to set the `DataType`. This works fine for the data fed into AES but KEY & IV are passed in in mode `CRYP_DATATYPE_32B` regardless of the setup of `DataType`.

see `static void CRYP_SetIV(CRYP_HandleTypeDef *hcryp)` & `static void CRYP_SetKey(CRYP_HandleTypeDef *hcryp, uint32_t KeySize)`.

This is especially problematic since key & iv are not passed to the hardware registers during `HAL_CRYP_Init` or `HAL_CRYP_SetConfig` but when calling `HAL_CRYP_Encrypt` which means we can't use local (stack) variables to correct the data alignment ourselves.

Please update `CRYP_SetIV` & `CRYP_SetKey` so they can deal with all modes of `DataType`.

Thanks,
Manuel

RJMSTM, 2024-03-04:
Hello @fermentedfly,

Thank you for reporting. Would you please clarify your problem do you want to have the ability to changes `DataType` into the Key and initialization vector.

Otherwise, you should first call `HAL_CRYP_SetConfig  `then `HAL_CRYP_Init `and then `HAL_CRYP_Encrypt `

Regards,

RJMSTM, 2024-03-22:
Hello,

In fact, the key and IV cannot be transmitted in little-endian format; they must always be sent in a one direction.

Regards,

fermentedfly, 2024-03-24:
Yes, because you designed the HAL driver in this way.

fermentedfly, 2024-03-25:
Correct, 
This was your design decision.
Thus the request to add the option to feed key & IV in little-endian.

RJMSTM, 2024-03-29:
Hello @fermentedfly,

I apologize, but there is no support planned for the 'little-endian' format for key and IV.

Please allow me to close the issue, thanks again for your contribution
 
Regards,
Rania

closed 2024-03-29 by RJMSTM