STMicroelectronics/STM32CubeU5 #43: STM32CubeU5  - SBSFU application cleanup required

tdjastrzebski, 2024-03-10:
[STM32CubeU5 SBSFU](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/SBSFU) app probably requires some cleanup. It contains lots of files completely unrelated to this solution: DSP, RTOS, Zephyr, DAP, NN (neural networks), several Templates folders and *_template.c files - to name just a few.

ALABSTM, 2024-03-15:
Hi @tdjastrzebski,

I had a look inside the different folders of the application and found none of the unrelated files you are mentioning. I also opened both `EWARM` and `STM32CubeIDE` project workspaces and found none included neither.

* The first screenshot below shows the [.../SBSFU_Appli/STM32CubeIDE](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/SBSFU/SBSFU_Appli/STM32CubeIDE) project workspace.
* The second screenshot shows the [.../SBSFU_Loader/EWARM](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/SBSFU/SBSFU_Loader/EWARM) project workspace.

Would you mind precising where those unrelated files could be found so we can do the cleanup? Thank you in advance for the clarification.

With regards,

[image]

[image]

tdjastrzebski, 2024-03-18:
It was my mistake, I included all the content of Drivers folder while only a small subset is needed.

ALABSTM, 2024-03-22:
Hi @tdjastrzebski,

You're welcome.

With regards,

closed 2024-03-22 by ALABSTM