STMicroelectronics/STM32CubeU5 #46: SBSFU app imgtool bug

tdjastrzebski, 2024-04-16:
[STM32CubeU5 SBSFU](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/SBSFU) app (latest revision 1.5) comes with probably custom `imgtool` implementation, which has at least one additional `flash` command not found in the current original MCUboot implementation. Method description reads:
_modify bash /batch variable value with layout file content_
Essentially, this method allows to update shell script variable values based on macro-preprocessed output C file like this (attached) example:
```C
enum image_attributes
{
    RE_IMAGE_FLASH_NON_SECURE_IMAGE_SIZE = (((((0x200000) + (0x200000)) - ((((((((((((0x0) + (0x2000)) + (0x2000)) + (0x10000))+(0x2000)) + (0x12000)) + (0x2000)) + (0x2000)) + (0x6000)) + (0x2000)) + ((0x6000))) + (0x2000)))))
};
```
After running this command:
`./Middlewares/mcuboot/scripts/dist/imgtool/imgtool.exe flash --layout ./<my_path>/test.c -b image_ns_size -m RE_IMAGE_FLASH_NON_SECURE_IMAGE_SIZE -s 0 ./test.sh`
**image_ns_size** variable value in **test.sh** script should get updated to **0x3C4000** value but it does not.
Instead, **0x43c000** value gets written.

Am I missing something? Test files attached. [test.zip](https://github.com/STMicroelectronics/STM32CubeU5/files/14996850/test.zip)

It seems that the current implementation does not always correctly subtract values.
There are other problems with this tool. In case the specified file path does not exist, non-specific _"local variable 'value' referenced before assignment"_ error message is displayed.
Unlike the output file path, input file path must be absolute.

Besides, why command modifying scripts is named **flash**? Perhaps a better name could be found.

[image]

ALABSTM, 2024-04-19:
Hi @tdjastrzebski,

Issue forwarded to our development teams. I will keep you informed. Thank you for having reported.

With regards,

ALABSTM, 2024-04-19:
ST Internal Reference: 179487

TOUNSTM, 2024-09-04:
Fixed in 6591004477fdd8fba04573c0373679f2be0da595

closed 2024-09-04 by TOUNSTM