STMicroelectronics/STM32CubeU5 #48: CMSIS layer never free()'s memory associated with a thread that has been detached and terminated

RobMeades, 2024-04-20:
We are using ThreadX through the CMSIS2 layer provided here with `USE_DYNAMIC_MEMORY_ALLOCATION`.

When we delete a task, we call [osThreadDetach()](https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c#L1507) and then [osThreadTerminate()](https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c#L1446), expecting this to free the memory that was allocated when we called [osThreadNew()](https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c#L702); however this is never the case, and we soon run out of memory.

You can see this if you set a break-point here:

https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c#L1458

...and here:

https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c#L1485

The second break point is never reached.  This is presumably because the thread has been terminated, task-switching has occurred, the code that follows is never going to be run.

How is this meant to work?

This tested on a Nucleo STM32F575ZI board, in case it matters.

ALABSTM, 2024-05-03:
ST Internal Reference: 180641

ALABSTM, 2024-05-03:
Hi @RobMeades,

Please excuse this delayed reply. Thank you for the detailed report. It has been forwarded to our development teams. We will try our best to get back to you by next week.

With regards,

ALABSTM, 2024-05-06:
Hi @RobMeades,

Our development teams acknowledged the issue and said they are already aware of it. They are working on a solution. Unfortunately, they still need time and it will probably not be for this week. They cannot share a date for the moment.

I will keep you informed should there be any update. We apologize for the inconvenience and count on your patience and comprehension.

With regards,

RobMeades, 2024-05-06:
@ALABSTM: understood, thanks for the update; as soon as you have a rough date let me know and I will let our customers know.

RobMeades, 2024-06-06:
Any update on this one (one of our customers is asking)?