STMicroelectronics/STM32CubeU5 #49: Ux_Device_CDC_ACM example Hardfault on high load

HiFiPhile, 2024-04-23:
**Describe the set-up**

SDK Version: https://github.com/STMicroelectronics/STM32CubeU5/commit/f0f063146b444e37ef6b1b3d2f219df728a12407
Board: NUCLEO-U5A5ZJ-Q
Compiler: IAR 9.50.1

**Describe the bug (skip if none)**

I'm a [TinyUSB](https://github.com/hathach/tinyusb) project maintainer, my planning was to compare the CDC throughput performance between usbx with a modified `Ux_Device_CDC_ACM` echo example.

But even without modification `Ux_Device_CDC_ACM` runs into hardfault on high load, `tx_byte_pool_buffer` overflowed and destroy all variables around. If I increase TX_APP_MEM_POOL_SIZE from 1k to 8k it crashes little later.

With attached serial testing script, crash happens about after 10 OUT transfers:

[image]

**How to reproduce the bug (skip if none)**

1. Compile `Ux_Device_CDC_ACM` project.
2. Use attached script, eg. `ser_test.py COM4`

[ser_test.py.txt](https://github.com/STMicroelectronics/STM32CubeU5/files/15075580/ser_test.py.txt)

ALABSTM, 2024-05-02:
Hello @HiFiPhile,

Thank you for this report. It will be forwarded to our development teams. I will keep you informed.

With regards,

ALABSTM, 2024-05-02:
ST Internal Reference: 180549

ALABSTM, 2024-05-07:
Hi @HiFiPhile,

Our development teams acknowledged what you reported. As a workaround, their advice is to set the request data size to be **512** (HS MPS) instead of **64** when calling `ux_device_class_cdc_acm_read()` while keeping actual data size **less than or equal to** 64.

https://github.com/STMicroelectronics/STM32CubeU5/blob/d52f12efe8c45c9efd629731c6eb396e8e7f7cdf/Projects/NUCLEO-U5A5ZJ-Q/Applications/USBX/Ux_Device_CDC_ACM/USBX/App/ux_device_cdc_acm.c#L286-L288

With regards,

HiFiPhile, 2024-05-07:
> while keeping actual data size less than or equal to 64.

Hi, it means each data packet sent from host should be less than or equal to 64 ?

ALABSTM, 2024-05-07:
Hi @HiFiPhile,

> Hi, it means each data packet sent from host should be less than or equal to 64 ?

Yes. That's the idea.

With regards,