STMicroelectronics/STM32CubeU5 #50: Incorrect defaults for UX_SLAVE_REQUEST_DATA_MAX_LENGTH

UnguidedFreedom, 2024-04-23:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**

 * Custom board using STM32U575VGT6
 * Project created using STM32CubeMX 6.11.0, using STM32CubeU5 v1.15.0

**Describe the bug (skip if none)**

When setting up USBX in STM32CubeMX with STM32CubeU5 in MSC mode, if I set `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` to any value other than 4096, then there is a `#define` directive in `USBX/App/ux_user.h`, however if I set it to exactly 4096, that line is commented out.

The problem is that [Middlewares/ST/usbx/ports/generic/inc/ux_port.h](https://github.com/STMicroelectronics/STM32CubeU5/blob/main/Middlewares/ST/usbx/ports/generic/inc/ux_port.h#L200) has code that defaults `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` to 2048 if it isn't set.

Putting those two things together means that if I set `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` to 4096 in STM32CubeMX, the actual value in my code will be 2048. 

**How to reproduce the bug (skip if none)**

1. Create a project in STM32CubeMX using STM32CubeU5
2. Enable USBX for the project in slave MSC mode
3. Set `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` to 4096 in STM32CubeMX
4. Generate sources
5. Run the project
6. Notice that the actual value of `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` is 2048 despite what was set in STM32CubeMX

**Additional context**

I was able to find a workaround by defining a custom constant `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` for USBX (which creates the constant in `main.h`), then update `USBX/App/ux_user.h` to add `#include "main.h"` in the `USER CODE BEGIN 1` part.

**Screenshots**

[image]
[image]
[image]

ALABSTM, 2024-04-29:
Hi @UnguidedFreedom,

Thank you for having reported. The behavior you reported could be reproduced. Our development teams will be notified. I will keep you informed.

With regards,

ALABSTM, 2024-04-29:
ST Internal Reference: 180337

ALABSTM, 2024-04-29:
Hi @UnguidedFreedom,

By the way, constant `UX_SLAVE_REQUEST_DATA_MAX_LENGTH` is commented out in one of the many applications provided within the STM32CubeU5 firmware, as you can see below. Let's wait for our development teams' feedback.

https://github.com/STMicroelectronics/STM32CubeU5/blob/f0f063146b444e37ef6b1b3d2f219df728a12407/Projects/B-U585I-IOT02A/Applications/OpenBootloader/USBX/App/ux_user.h#L133

With regards,

ALABSTM, 2024-05-07:
Hi @UnguidedFreedom,

Our development teams acknowledged the issue. A fix will be available in the frame of `STM32CubeMX` release **6.12.0**. As there is no impact on the software provided within this repository, please allow me to close this issue.

Thank you for your comprehension and thank you again for having reported.

With regards,

closed 2024-05-07 by ALABSTM