STMicroelectronics/STM32CubeU5 #53: NUCLEO-U5A5ZJ-Q HSE clock configuration

builder60, 2024-10-01:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**

STM32CubeMX Version 6.12.1
STM32Cube_FW_U5_V1.6.0

Start a new project from the ST Board Selector and select NUCLEO-U5A5ZJ-Q.
The default configuration for HSE is Bypass and 8Mhz. This will not work on the U5A5 board as the board is configured with a 16Mhz Crystal.

**How to reproduce the bug (skip if none)**

1. Indicate the global behavior of your application project
2. List the modules that you suspect to be the cause of the problem (Drivers, BSP, MW...)
3. Describe the use case that generates the problem
4. How we can reproduce the problem

**Additional context**

If you have a first analysis, an enhancement, a fix or a patch, thank you to share your proposal.

**Screenshots**

If applicable, add screenshots to help explain your problem.

KRASTM, 2024-10-03:
Hello @builder60,

Thank you for the report.

Regarding the issue, and based the [RM0456](https://www.st.com/resource/en/reference_manual/rm0456-stm32u5-series-armbased-32bit-mcus-stmicroelectronics.pdf) page 484, when choosing the External source HSE (Bypass), in this mode an external source must be provided to the OSC_IN pin while the OSC_OUT pin can be used a GPIO.

In the case of NUCLEO-U5A5ZJ-Q board and according to the [UM2861](https://www.st.com/resource/en/user_manual/um2861-stm32u5-nucleo144-board-mb1549-stmicroelectronics.pdf), page 27: **The MCO output of the STLINK-V3E MCU is used as an input clock. The frequency might be changed during the ST-LINK firmware upgrade (stsw-link007). It is fixed at 8 MHz by default and connected to PH0 OSC_IN of the STM32 microcontroller**.

That's why when you start a new project with MX, `the default configuration for HSE (Bypass) is 8Mhz`. Remember that you can choose the HSE crystal as clock source if you want.

Please, allow me to close the issue, thank you for your comprehension.

With regards,

builder60, 2024-10-03:
The NUCLEO-U5A5ZJ-Q that I purchased was configured by ST as in the paragraph above the paragraph that you quoted:

**HSE: on-board oscillator crystal (X3) (configuration depends on the embedded STM32U5 series
microcontroller)**
To help select the crystal and its associated capacitors (C59, C60), refer to the application note Oscillator design guide for STM8AF/AL/S, STM32 MCUs and MPUs (AN2867).
As an example, this crystal might have the following characteristics: 16 MHz, 8 pF, and 20 ppm.
In this case, configure the board as follows:
• SB47 and SB49 OFF. PH0 and PH1 are not connected to CN11 as I/Os.
• SB48 (MCO) OFF
• SB4 and SB5 ON are connected to the external HSE.
[image]

UM2861 is not correct about the default clock source.

KRASTM, 2024-10-03:
Hello @builder60,

Indeed, the SB4 and SB5 is ON.

I will check with our team about the configuration within Cube MX.

Thank you again.

KRASTM, 2024-10-03:
ST Internal Reference: 192857

KRASTM, 2024-10-11:
Hello @builder60,

Regarding the issue, the fix will be included in a future release of Cube MX. Please allow me to close the thread.
Thank you for your comprehension.

With regards,

closed 2024-10-11 by builder60