STMicroelectronics/STM32CubeU5 #55: Uninitialised variables in stm32u5xx_hal_dma_ex.c cause compile failure

rleigh-electradx, 2025-01-05:
**Setup**

Board: Nucleo U575
Compiler: [Arm GNU Toolchain 14.2Rel1](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)
Compiler options detailed below, but the important ones are `-Wall -Werror`

**Bug description**

It looks like there are at least two instances of uninitialised structures which result in compile failure when warnings are being treated as errors.  In this case, it looks like these are outright bugs--I can't see these initialised in the codepaths the compiler is highlighting.  It should be possible to compile in this way if the variables are correctly initialised for all possible codepaths.

```
GCC 14.2.1
Options: -DSTM32U575xx -DSTM32_THREAD_SAFE_STRATEGY=2 -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -Iproject/Drivers/STM32U5xx_HAL_Driver/Inc -Iproject/Drivers/STM32U5xx_HAL_Driver/Inc/Legacy -Iproject/Core/Inc -Iproject/Drivers/CMSIS/Device/ST/STM32U5xx/Include -Iproject/Drivers/CMSIS/Include -O3 -DNDEBUG -std=gnu11 -fdiagnostics-color=always -mcpu=cortex-m33 --specs=nano.specs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -mthumb-interwork -ffunction-sections -fdata-sections -fno-common -fmessage-length=0 -fstack-usage -Wall -Werror --specs=nosys.specs

project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_InsertQ':
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:2362:45: error: 'dest_q_node_info.nextnode_addr' may be used uninitialized [-Werror=maybe-uninitialized]
 2362 |             (dest_q_node_info.nextnode_addr & DMA_CLLR_LA) | cllr_mask;
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:2199:26: note: 'dest_q_node_info.nextnode_addr' was declared here
 2199 |   DMA_NodeInQInfoTypeDef dest_q_node_info;
      |                          ^~~~~~~~~~~~~~~~
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ConvertQToDynamic':
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:2980:77: error: 'node_info.currentnode_addr' may be used uninitialized [-Werror=maybe-uninitialized]
 2980 |     if (((uint32_t)pQList->FirstCircularNode != 0U)                         &&
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
 2981 |         ((uint32_t)pQList->FirstCircularNode != node_info.currentnode_addr) &&
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:2927:26: note: 'node_info.currentnode_addr' was declared here
 2927 |   DMA_NodeInQInfoTypeDef node_info;
      |                          ^~~~~~~~~
In function 'DMA_List_ConvertNodeToStatic',
    inlined from 'HAL_DMAEx_List_ConvertQToStatic' at project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:3076:5:
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:4492:89: error: 'context_node' may be used uninitialized [-Werror=maybe-uninitialized]
 4492 |       current_node->LinkRegisters[contextnode_reg_counter] = context_node->LinkRegisters[contextnode_reg_counter];
      |                                                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ConvertQToStatic':
project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c:3028:19: note: 'context_node' declared here
 3028 |   DMA_NodeTypeDef context_node;
      |                   ^~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

To reproduce, compile with `-Wall -Werror`.  The above failure should be reproduced.

rleigh-electradx, 2025-01-05:
I made these changes.  They get rid of the warnings, but will need checking for correctness in case I've masked any real bugs.

```diff
diff --git a/project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c b/project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c
index d60eca2..d19d115 100644
--- a/project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c
+++ b/project/Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma_ex.c
@@ -2196,7 +2196,7 @@ HAL_StatusTypeDef HAL_DMAEx_List_InsertQ(DMA_QListTypeDef *const pSrcQList,
   uint32_t cllr_mask;
   uint32_t cllr_offset;
   DMA_NodeInQInfoTypeDef src_q_node_info;
-  DMA_NodeInQInfoTypeDef dest_q_node_info;
+  DMA_NodeInQInfoTypeDef dest_q_node_info = {0};
 
   /* Check the source and destination queues and the previous node parameters */
   if ((pSrcQList == NULL) || (pDestQList == NULL))
@@ -2924,7 +2924,7 @@ HAL_StatusTypeDef HAL_DMAEx_List_ConvertQToDynamic(DMA_QListTypeDef *const pQLis
   uint32_t cllr_offset;
   uint32_t currentnode_addr;
   DMA_NodeTypeDef context_node;
-  DMA_NodeInQInfoTypeDef node_info;
+  DMA_NodeInQInfoTypeDef node_info = {0};
 
   /* Check the queue parameter */
   if (pQList == NULL)
@@ -3016,6 +3016,12 @@ HAL_StatusTypeDef HAL_DMAEx_List_ConvertQToDynamic(DMA_QListTypeDef *const pQLis
   return HAL_OK;
 }
 
+#if defined(__GNUC__) && !defined(__clang__) && !defined(__llvm__)
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wuninitialized"
+#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"
+#endif
+
 /**
   * @brief  Convert a linked-list queue to static (Not optimized DMA queue execution).
   * @param  pQList : Pointer to a DMA_QListTypeDef structure that contains queue information.
@@ -3025,7 +3031,7 @@ HAL_StatusTypeDef HAL_DMAEx_List_ConvertQToStatic(DMA_QListTypeDef *const pQList
 {
   uint32_t cllr_offset;
   uint32_t currentnode_addr;
-  DMA_NodeTypeDef context_node;
+  DMA_NodeTypeDef context_node = {{0},0};
 
   /* Check the queue parameter */
   if (pQList == NULL)
@@ -3091,6 +3097,10 @@ HAL_StatusTypeDef HAL_DMAEx_List_ConvertQToStatic(DMA_QListTypeDef *const pQList
   return HAL_OK;
 }
 
+#if defined(__GNUC__) && !defined(__clang__) && !defined(__llvm__)
+#pragma GCC diagnostic pop
+#endif
+
 /**
   * @brief  Link linked-list queue to a DMA channel.
   * @param  hdma   : Pointer to a DMA_HandleTypeDef structure that contains the configuration information for the
diff --git a/Threads/Console/Console.cpp b/Threads/Console/Console.cpp
index 5aaa9a0..01f613e 100644
--- a/Threads/Console/Console.cpp
+++ b/Threads/Console/Console.cpp
@@ -200,6 +200,7 @@ void console_thread_entry(ULONG thread_input) {
     HAL_UART_Transmit(&CONSOLE_UART, reinterpret_cast<const uint8_t *>(&prompt[0]), strlen(prompt), 100);
     auto dma_status = HAL_UARTEx_ReceiveToIdle_DMA(&CONSOLE_UART, console_rx_buffer.data(),
                                  console_rx_buffer.size());
+    (void)dma_status;
     assert(dma_status == HAL_OK);
 #endif
```

TOUNSTM, 2025-01-08:
Hello @rleigh-electradx,

Thank you for  your report. We have attempted to reproduce the compilation errors you mentioned but were unable to do so. It is important to note that the **STM32U5** drivers currently support GCC12, and at this time, we do not have plans to support GCC14. Please allow me to close this issue.

Best Regards,

closed 2025-01-08 by TOUNSTM